MarekKowalski/LiveScan3D #6: How can i use your server application in C++

kirubhakinect, 2015-12-16:
Currently i am trying to pass kinectv2 streams between different machines.I like your way of sending and receiving kinect streams between different machines very much.But the only thing is your client application is in C++ and server application is in C#.

Is there any source which you have developed server application in C++.

Please suggest me how can i use your server application in C++.

Thanks ,
Kirubha

MarekKowalski, 2015-12-16:
Hi,

Unfortunately I do not have any version in C++. However if you only want to send and receive streams (not settings windows, live display etc.), creating a C++ version should be easy.
I suggest you start with the socket library that I use in KinectClient, as it is very easy to use. 

In order to have a "minimal" setup working, you need to reimplement the KinectSocket and KinectServer classes. Or at least their basing functionality which is, accepting connections from clients and requesting/receiving frames.

Let me know if you need more help with this.

Best,

Marek

kirubhakinect, 2015-12-16:
Thanks Marek for you immediate reply.Yes i am also looking for client server code using socket library.So that i could try to send and receive kinect streams between machines.

Please tell me any link that would do server code using socket library that would be much more helpful to me.

Thanks,
Kirubha

MarekKowalski, 2015-12-17:
The socket library I use in the KinectClient has a very simple server example that should get you started here:
https://github.com/ReneNyffenegger/development_misc/blob/master/c%2B%2B/socket/examples/EchoServer.cpp

Look at the ReceivingWorker method in KinectServer.cs, specifically the main loop of this method. Try and reimplement similar message handling in C++ based on the example I just sent you.

kirubhakinect, 2015-12-20:
Thanks Marek,
         I have created sample server application using the link what you given.But when i debugging the  RunProxyThread function.

  unsigned __stdcall RunProxyThread (void* a) {
  Socket*      s = (Socket*) a;
  SocketClient c(addrServer, portServer);
  while (1) {
    SocketSelect sel(s, &c);
   bool still_connected = true;
   if (sel.Readable(s)) {
    std::string bytes = s->ReceiveBytes();
     // c.SendBytes(bytes);
      std::cout << "Server: " << bytes << std::endl;
      if (bytes.empty()) still_connected=false;
    }
    if (sel.Readable(&c)) {
      std::string bytes = c.ReceiveBytes();
   //   s->SendBytes(bytes);
      std::cout << "Client: " << bytes << std::endl;
      if (bytes.empty()) still_connected=false;
    }
    if (! still_connected) {
      break;
    }
  }
 delete s;
return 0;
}

The SocketSelect readable API function returns byte data but we cannot identify whether the function returns color,depth,body data. I am getting the bytes like this "250,510,3435973836,3435973836,3435973836" and the buffer length also 64 only.At the sametime ReceiveBytes function does not returns any data.Please help me to read entire data from client side with separate buffer for color,depth and body data.

Thanks,
Kirubha

MarekKowalski, 2015-12-21:
Hello,

The ReceivingWorker function in the KinectServer shows you how to distinguish between different types of messages. In short: if the first byte of a message is 2 or 3, then that message is a frame. A frame will contain depth, color and body data. In order to see how frames are read, look at the KinectSocket::ReceiveFrame method. You can also look at how they are sent in the client.

Also - please notice that in order to receive any frames, you first need to request them by sending a byte equal 4 to the client. Looking at how the client (which is in C++) handles the messages might help you.

I suspect that the bytes you mention might not be the bytes you receive. Take a moment to look at their values: "250,510,3435973836,3435973836,3435973836". Only the first one is within the range of a byte, which is 0 - 255.

Best regards

Marek

kirubhakinect, 2016-01-03:
Thanks marek,Your answer helped me a lot to identify color,depth and body data.Now i can see point clouds coming from another machine.It really gives me great joy to see another machine data from my server machine.Once again a great thanks to you.

I have a few questions running on my mind.Please suggest me with answer for this questions.

1.I want to display the other machine point cloud with in 512 * 424 resolution.How can i do that?

2.How to get depth data from livescan application?

3.How can i map any skeletal joint cameraspace to depth space?

4.How to map depth point from your 3D point cloud?

Thanks,
Kirubha

MarekKowalski, 2016-01-07:
Hello,

I am glad to hear that you have made progress. 

1. You cannot really display a point cloud at any resultion. What I think you meant is displaying the point cloud as depth data at the given resolution, which brings us to your second question.

2. I understand that you want to obtain separate depth maps from each of the Kinects you are connecting to the server. In order to do that I suggest you modify the client application so that it sends the depth maps to the sever instead of the point clouds. There is no easy way to project the point clouds to depth maps on the server side as the projection parameters of the Kinects are not known (or at least I do not know how to extract them from the CoordinateMapper class).

3 and 4. You can do it easily on the client side using the CoordinateMapper class, read more about it here:
https://msdn.microsoft.com/en-us/library/dn785530.aspx?f=255&MSPPError=-2147217396

Best regards

Marek

kirubhakinect, 2016-01-11:
Hi,

Yes great thanks to you marek,

Following are my next task:-

1.My next task is to send depth from client machine to server machine.Please tell me in your client code, which part i need to concentrate in order to send depth to server.

2.I need  floor clip plane values of client machine and send it to server.

Please suggest me with some ideas to proceed.

Thanks,
Kirubha

MarekKowalski, 2016-01-11:
Hi Kirubha,

1. The point cloud is being sent in the "SendFrame" function in liveScanClient.cpp.
The depth data is store in pCapture which is a pointer to ICapture in the main file. Inside ICapture there is a field called pDepth, which is an array containing the depth data. The size of the array is nDepthFrameHeight * nDepthFrameWidth.

2. The floor clip plane is a part of the BodyFrameIndex object, as you can see in the SDK documentation, here:
https://msdn.microsoft.com/en-us/library/windowspreview.kinect.bodyframe.floorclipplane.aspx
The BodyFrameIndex data is read using the KinectCapture class, which is derived from ICapture (pCapture is an object of that class). I suggest that you add the floor clip plane field to ICapture and read it in KinectCapture. If you do that you will be able to access and send it easily inside the main class.

Best regards and good luck,

Marek

kirubhakinect, 2016-01-12:
Thanks Marek,Your explanations are very clear.Now i know how to start getting depth data and floor clip plane.

closed 2016-04-19 by MarekKowalski