MarekKowalski/LiveScan3D #8: Linux makefiles and docs

deniskulicek, 2016-02-12:
You mentioned that this also runs on Linux, and that it is possible to have multiple clients only when using Linux. Is it possible to make available makefiles for building this on linux?

I managed to build LiveScanServer using MonoDevelop, however it crashes when started.

Thanks!

MarekKowalski, 2016-02-12:
Hi Denis,

What I must have mentioned was that I know that it is possible to use multiple Kinects on one computer on Linux. You can do it using the drivers available here: https://github.com/OpenKinect/libfreenect2. The only problem is that non-official drivers do not have support for skeleton tracking (and maybe some of the other functionalities, I am not sure).

I do not think it is possible to run my app on Linux as it uses the Kinect for Windows library, which I imagine is only available on Windows.

Best regards!

Marek

deniskulicek, 2016-02-12:
Thanks Marek,

I must have misinterpreted your note about running multiple Kinects on Linux as running multiple instances of LiveScanClient.

Best regards,
Denis

closed 2016-02-12 by deniskulicek

deniskulicek, 2016-02-12:
Btw Marek,

Are you aware of similar projects that run on Linux too? Otherwise, how would you recommend porting this one to Linux? I guess port to libfreenect2 would be first in the list - skeleton tracking is something I can live without so that's not an issue.

Thanks,
Denis

MarekKowalski, 2016-02-15:
Hi Denis,

Unfortunatelly I do not know any similar projects that are open source. 

I don't know much about porting C# code to Linux so I can't help you with that part.
I have recently heard that it is possible to use multiple Kinects on one computer under Windows using the same library as for Linux (libfreenect2). If you got that library to work on Windows, than I image it would be much easier to work with the C# code there.

As for using libfreenect2 with the client app, it should be quite easy.
You would have to implement the iCapture interface similarly to how it is implemented for the Kinect SDK in the KinectCapture class. Since there would be no body data you would just leave the body vector empty. 

Best regards,

Marek