MarekKowalski/LiveScan3D #10: Unable to get client to run from source

wiegerthefarmer, 2016-04-18:
I need to make some changes to the client. I've been able to compile it, and it actually seems to run just fine. Connects to the Kinect, and I can connect to the server and send data across.

However, if I run the calibrate command from the server to the client, the client crashes on
MarkerDetector::GetMarker
...
cv::findContours(img3, contours, CV_RETR_CCOMP, CV_CHAIN_APPROX_NONE);

I get this exception

_> Exception thrown at 0x0F6A0770 (opencv_core248.dll) in LiveScanClientD.exe: 0xC0000005: Access violation reading location 0x0406A0DC.
> 
> If there is a handler for this exception, the program may be safely continued._

Any thoughts?

If I just run the included client, everything works fine.

MarekKowalski, 2016-04-19:
Hello Aaron,

Which Visual Studio version are you using?
The binary was compiled with Visual Studio 2013, I believe that if you are using a different version of visual studio, you might need different OpenCV dll files.
Are you using the dlls from the binary version from my website?

Best regards

Marek

youngheekim, 2016-04-25:
I got a similar problem when i run the calibrate command from source. The binary version works fine in my system(one server computer and one client computer). 

I am working with Visual Studio 2013 and the same version of OpenCV dll files(the binary version from your website is using OpenCV2.4.8).

The client crashes on 
bool Calibration::GetMarkerCorners3D(vector<Point3f> &marker3D.....) 
{ ... 
float dx = marker.corners[i].X - minX;
float dy = marker.corners[i].Y - minY;
... }
since the function GetIntersection() returns strange numbers on  
void MarkerDetector::CornersSubPix(vector<cv::Point2f> &corners,...) 
{
	vector<cv::Point2f> corners2;
	for (unsigned int i = corners.size() - 1; i < 2 * corners.size() - 1; i++)
		corners2.push_back(GetIntersection(lines[(i + 1) % corners.size()], lines[i % corners.size()]));
corners = corners2;
}

Any thoughts?

 Best regards

MarekKowalski, 2016-04-27:
Hi,

I can't get this error on my computer, I'll try on a couple of other computer tomorrow.
Are you using Windows 10 or 8?

Marek

MarekKowalski, 2016-04-28:
Hi,

So I am not really able to get those errors on my side on my or any other computer. Are you using Debug or Release mode?
When GetIntersection returns strange results, how do they look like, is it something like Inf or Nan?
Could you try to see if it works if you comment out the call to CornersSubPix in GetMarker?

Thanks a lot,

Marek

youngheekim, 2016-04-29:
Hi,
There was no crash after commenting out the call to CornersSubPix(). Calibration works fine and the result of "Show live" looks good.  
Could you explain what the CornersSubPix() is?  

By the way, I am using Windows 8.1 and tested in Debug and Release mode.

Thanks a lot. 

Young.

MarekKowalski, 2016-05-02:
Hi,

I'm glad that it helped, but I'm still puzzled at why that method wouldn't work :(
I will take a closer look at CornersSubPix when I have a moment and try to determine what the reason might have been.

CurnersSubPix fits lines to points located at each side of the detected markers and then finds the intersections of those lines. This provides the locations of marker corners with sub pixel precision, and should improve accuracy of calibration.

Marek

closed 2016-05-02 by MarekKowalski