MarekKowalski/LiveScan3D #16: Very slow

CreateOneMore, 2017-06-01:
Hello, when I use LiveScan3D with my PC, the status bar show the FPS just 0.5 - 2. How can I make it more  quickly? I have set Stream only bodies, and disable filtering. The client and server is run on the same PC.

MarekKowalski, 2017-06-01:
Hi,

Are you running a precompiled binary from the release or did you compile it yourself?
If you compile it yourself did you use the Release settings (as opposed to Debug which is slower)?
Is this the framerate from the client or the live view window in the server?

Marek

CreateOneMore, 2017-06-02:
I try to compile it use the Release settings today. It looks good. Is anything difference between Debug and Release? And my PC can't run the binary that download at release page, i had double click the EXE file several times but nothing happened, Even if i use the administrator permission.

closed 2017-06-02 by CreateOneMore

MarekKowalski, 2017-06-02:
In release the program is compiled with optimization while in Debug it is not, that is where the performance improvement comes from. I am sad to hear the binary did not work on your machine, I will look into it. 

Best regards,

Marek