MarekKowalski/LiveScan3D #17: Connecting to a VM on the Azure Cloud

gotschallrw, 2017-06-01:
I'm trying to connect the LiveScanClient to a server that is running within a VM hosted in the cloud with a public IP Address.  When I attempt to connect from the LiveScanClient I get a "Failed to connect. Did you start the server?" message.  The VM is running Windows Server 2008 and has Windows Firewall configured to allow any messages through to the LiveScanServer application.

Is connecting to an outside IP Address currently supported?  Are there any specific requirements for running the Server application?

MarekKowalski, 2017-06-02:
Hi, did you make sure that port forwarding for port 48001 is enabled?
LiveScan uses that port for communication. I have not used Azure before but I know that on AWS you would need to enable port forwarding through that port for the connection to be established. 

Let me know if that solved it.

Best regards 

Marek

MarekKowalski, 2017-06-12:
Closing this issue, since the original poster is not responding.

closed 2017-06-12 by MarekKowalski