MarekKowalski/LiveScan3D #18: can not calibrate

zhao-zilong, 2017-06-12:
when I try to calibrate the kinect, it always shows a green border, but I already add the marker Id, and from the part of client, the video that kinect captured is mirror symmetry, I just blocked here.

MarekKowalski, 2017-06-12:
Hi,

The video being mirrored is normal.
Have you tried to move the Kinect around to see whether it starts working then?
Can you send a screenshot of your settings window?
Also, is it the newest version of the app or the one from the Release?

Best regards,

Marek

zhao-zilong, 2017-06-12:
Hi,

Finally,  using the exe file of LiveScanClient and LiveScanServer solved this problem.
Thank you Marek for your patients

Best

ZHAO

closed 2017-06-12 by zhao-zilong