MarekKowalski/LiveScan3D #19: i am trying the app, but have issues

hbzhang, 2017-11-22:
The live view is simply the dark screen, nothing in it.  I tried using kinect SDK body 2D, it can capture my skeleton normally.  I noticed that when the Livescan server started, Kinect camera LED light (white color) is not activated. When using Kinect SDK, when the camera is connected, the camera LED light is activated. 

Also, do I need to specify server IP for the player?

Do I need marker to calibrate kinect in order to use server?  What is the bounding box means here?

Thanks guys for this amazing job!!

MarekKowalski, 2017-11-22:
Hi!

Do you start both the client and the server? When you start the client, is there any message on the status bar saying why the Kinect did not start?

The marker commes into use later, once you can connect the client to the server. The bounding box is the area of the scene that will be reconstructed, it will be pretty obvious once everything starts working.

Marek

hbzhang, 2017-11-22:
Thanks Marek for the quick reply. I started player (which has Kinect V2). I also started server on another machine (no Kinect v2). Both show live are dark screen. 

When I click on start player, there is nothing show on the player's status bar.

hbzhang, 2017-11-22:
One thing strange is that if the player works, then the LED light (indicator) on Kinect V2 should be activated. But apparently after I clicked the start player, the Kinect V2 LED not activated at all.

MarekKowalski, 2017-11-23:
Hi,

As I mentioned before you need to start the **client** not the player.
The player is for playing prerecorded sequences. It does not run the Kinect, that's why the light is not on.

Let me know if that works.

Marek

hbzhang, 2017-11-23:
Thanks. sorry. I open the livescan3d client and it works now. 

Now, I am trying to connect it with hololens. 

After I enter the server IP on hololens, nothing shows up on HoloLens - completely blank after I enter the IP. I follow the instruction and open the live view window on livescan3d server. I also enter the server IP on HoloLens.

Is there something I am missing here?

thanks!

MarekKowalski, 2017-11-24:
Hi!

I know that this one might be obvious, but have you look around the room (including behind you) to see if the reconstructed point cloud is there? This was an issue with several people before.

If not I recommend you try two things:
 - run the HoloLens application in Debug mode and see if receives data,
 - rebuild the NetworkCommunication library and copy the resulting .dll and .winmd files to the Assets foder in the Unity project, then go to Unity and build the VS solution again.

Let me know what the issue was and if it ends up working. Hearing about those issues is very helpful later on.

Thanks,

Marek

blackghost1987, 2017-11-24:
Also check the server settings to see if "Stream only bodies" option is enabled. If it's enabled the server will only stream a part of the point cloud associated with a recognized skeleton, and won't stream the whole room behind it. If there's no recognized skeleton, there will be nothing to see in the HoloLens, so you have to disable it or have an actual person in the scene. Note that if this is enabled it makes the performance a lot better, as the point cloud to stream and render is much smaller.

hbzhang, 2017-11-27:
Thanks all for the comment. Sorry for the delay due to family issues. I am testing it now. I can update it more later.