MarekKowalski/LiveScan3D #20: Multiple Marker Calibration Problem

Lecanyu, 2017-12-01:
Hi, 

Recently, I have tried to calibrate three Kinect2 which were put around a small office, their position look like an equilateral triangle. 

Obviously, It is impossible to use only one marker to calibration all of Kinect2. So I have to use 2 marker. But After added 2 calibration marker in LiveScanServer Setting, I cannot get correct calibration of three Kinect2.

I noticed that there are Orientation and Translation slots in Calibration Marker Setting. Dose it mean I should provide the transformation between different Maker?

BTW, If I use the default Orientation and Translation (i.e all of six slots are 0), the calibration process looks like ambiguous because two different marker simultaneously locate global coordinate origin.

Could you tell me what the standard procedure of multiple marker calibration is? I cannot find this information from manual.

Lecanyu, 2017-12-08:
Thank you for your reply!

For calibration part, I had gotten that from your paper.  

But as similar with your second paragraph saying, Kinect scanning point cloud seems existing larger intrinsic error. Right? 

Because the drift became bigger in rigid box area after ICP refine. (The rigid box area stitches better after only initial calibration without ICP refine).

MarekKowalski, 2017-12-08:
Hi,

I'm sorry but I do not understand your question fully?
Are you saying the error is higher after the ICP refinement?
That might happen for some scenes, as I suggested you might want to try to limit the bounding box.
If that does not work I suggest you do the refinement when a more "complex" object is in the scene, for example a human.
ICP might converge to a bad solution if only a few planes are present.

Marek

Lecanyu, 2017-12-08:
Hi, 

Yes, I talked about error increasing after ICP refinement. 

But except local optima problem. What I want to ask is:

Is it possible that there is intrinsic error in Kinect scanning? _(i.e Foreground is aligned good after initial calibration but background is aligned bad. After ICP refinement, background is aligned good but foreground is aligned bad)_

MarekKowalski, 2017-12-11:
Hi,

Yes that is also possible. One of the sources of such error is the Multipath Intereference.

Marek

Lecanyu, 2017-12-11:
OK, I also got that interference from your paper.

Thanks for your reply!

closed 2017-12-11 by Lecanyu