MarekKowalski/LiveScan3D #21: Which sensor I should use to replace Kinect V2?

ChienDuong, 2018-01-03:
Recently, KinectV2 is discontinued to produce. So I change Kinect V2 to Kinect Xbox One. But today, the Kinect Xbox One is also discontinued to produce!. 
I am trying to find a replacement sensor which can be used in LiveScan3D. Can you give me a suggestion?

Thank you so much for this work!

Mark

earlvanze, 2018-01-03:
Kinect V2 IS Kinect for Xbox One. There are only two Kinect models. Also, there's tons of used ones available at GameStop and other used game stores for $16.

[Earl Co - Contact Using Hop](http://GetHop.com/?_hmid=1514991663)

ChienDuong, 2018-01-03:
Thank you earlvanze. This is a good option. But actually, I hope that the work can be used for a long time in the future. Thus, if I can find a replacement sensor or a long-term solution, it is much better.

earlvanze, 2018-01-03:
Hi Mark,

Just because Microsoft has stopped producing more Kinects and discontinued further hardware and game development doesn't mean you can't buy up the existing inventory and use it for a long time in the future. As long as LiveScan3D continues to be supported it wouldn't matter if that's all you're going to use the sensor for. Are you planning on playing any Xbox games with it?

Regards,

Earl

MarekKowalski, 2018-01-04:
Hi,

I agree that there are still plenty of Kinects on the market. Please note however that modifying the application to work with a new sensor should be very easy. All one has to do is implement one interface class in the LiveScanClient.

We are thinking about implementing interfaces for other sensors, such as Intel RealSensein, however there is never enough time to actually do it. If anyone is interested in implementing such an interface, it would be a very interesting addition to the application.

Best regards,

Marek

ChienDuong, 2018-01-05:
Hi Earl,
yeah, I am trying to control the 3D model base on the fused skeleton.

Hi Marek,
Thank you so much. I also think about the RealSense camera.

closed 2018-01-05 by ChienDuong