MarekKowalski/LiveScan3D #22: Machine configuration required for running the LiveScan3D server

aviban, 2018-01-15:
Hi Team, please let me know if it necessary to have a GPU for running the server? I don't have GPU in my current machine and whenever I clicked on the "Show Live" button, the LiveScan3D window is going into a "Not responding" state.

MarekKowalski, 2018-01-15:
Hi,

I think you need a GPU to display the live view window, but other than that, it should be fine without it.

Marek

aviban, 2018-01-15:
Thanks Marek for your valuable advise. We are going to procure one GPU.

Thanks again for your quick reply.

MarekKowalski, 2018-01-15:
Hi,

Just to be clear, you do not need a General Purpose GPU, any GPU should be enough. Most current CPUs have a built in GPU, so on most machines there should be no need for a separate GPU.
There might however be issues if you want to run this on a machine that is in a cloud.

Marek