MarekKowalski/LiveScan3D #23: Problems with calibration

ismaelgsan, 2018-02-06:
Hi, we have problems with calibration of multiple Kinects because when we do the steps of the documentation, the result is two point clouds on different calibration. What is wrong?
Please, can you write us step by step how to do?
Thanks!!
[image]

MarekKowalski, 2018-02-09:
Hi,

How many markers are you using and how many did you specify in the settings menu?
If you specified multiple markers in settings, did you make sure that their relative poses are correct?

Marek

ismaelgsan, 2018-02-09:
Hi @MarekKowalski,

This morning we can fix our problem using a single marker.
We were using six markers because we have two Kinects with different poses and only one marker was captured in both Kinetcs. Our problem was the last marker are detecting by each Kinetc is not the same and we were using two in the settings menu.

Anyway, I'm not sure if I understand the calibration system. What happen when we have two markers one in front of the other one and we use two Kinects one in front of the other one? We have to specify both in the settings menu? We should use the calibration file?

Thank you so much for reply.

Regards,
Ismael.

MarekKowalski, 2018-02-12:
Hi,

A couple of clarifications:
 - Each Kinect will only use the largest marker it can see.
 - If the largest marker seen by a Kinect is not specified in the settings menu, it will not calibrate.
 - Each marker you use has to be specified in the settings menu along with its position and orientation.
 - The Kinect determines its position relative to the marker, and subsequently uses the marker's position specified in the settings to establish its position in the world coordinate system. This means that if you do not specify the marker's position in settings correctly, it will not calibrate correctly.
 
Does that clarify all of your concerns?

Best regards,

ismaelgsan, 2018-02-12:
Hi,

I understand you. Really it's simple with your clarifications. Everything work fine.
I only want thank you for that and congrat you for this incredible project. It's awesome!

Regards,
Ismael.