MarekKowalski/LiveScan3D #24: problems with server

torpor29, 2018-06-04:
I can't open the show live window if I use the released server. But if I just click the debug button to run the server, I can open the show live. How to solve the problem?

MarekKowalski, 2018-06-05:
Hi,

Are you running the application with the debugger attached? Please try running it without the debugger (Ctrl+F5). If that does not work, I suggest you try the release available here:
https://github.com/MarekKowalski/LiveScan3D/releases

Thanks,

Marek

songqiming97, 2023-07-07:
[image]
Hello, I want to ask you, I used two Liveclent ports to connect the server, but when I clicked showlive, only one Kinect could be displayed, and no image could be displayed