MarekKowalski/LiveScan3D #25: client is not getting connected with the server

ShaminiKoravuna, 2018-06-08:
Hello,

I'am trying to use the application. Am able to run the LiveScanClient application successfully on two laptops . But I'am facing an issue while connecting the client with the server. Even thought after starting the server it's showing an error as Failed to connect, Did you start the server?

Could you please check with this?

Thanks a lot for such a great work!

Shamini

[image]

naruniec, 2018-06-08:
> whatsapp image 2018-06-08 at 13 41 45

Hi,
Thank for appretiation!
In the server application you have to push "Start server" button. Also 
make sure, that the server address in the client application is valid 
and that you don't have firewall or anti-virus preventing the connection.
Best,
Jacek

W dniu 08.06.2018 o 13:55, ShaminiKoravuna pisze:

ShaminiKoravuna, 2018-06-11:
Hello,
I have checked it with the things mentioned by you and even firewall or anti/virus is not preventing the connection. Still am facing the same problem. Could you please suggest any other alternative?
Regards,
Shamini

closed 2023-03-28 by ShaminiKoravuna