MarekKowalski/LiveScan3D #27: File format saving problem

18738753011, 2018-07-05:
File format saving problem
Hi, we have a question about saving files. When we choose Ascll_ply or binary_ply, the saved file format is .bin format, we want to save it as .ply format. Maybe because we may not really understand the meaning of the paper, so we have two questions to ask: First, whether the point cloud can be saved after multi-view registration, and second, whether the point cloud after multi-view registration can be saved .ply format?

MarekKowalski, 2018-07-09:
Hi,

Yes the data can be saved to .ply format.

During recording the .bin file is treated as a temporary buffer by the client.
Once the recording is finished, the clients send the recorded frames to the server, which subsequently saves them as .ply files in the directory specified in the main window of the server app.

Let me know if that answers your questions.

Thanks,

Marek

ChienDuong, 2018-08-06:
Hi Marek,

When Savingframes, Clients  read the frame which is recorded (which has points and colors). However, when Clients sends to sever, they send points + colors+ **m_vLastFrameBody**. Thus, the Skeleton will not synchronize with the points and colors right?

To solve the problem, maybe in WriteFrame, we also need to write the Bodies information, then read it and send to Sever. Am I right? 

Thanks,
Chien

MarekKowalski, 2018-08-14:
Hi Chien,

That's a good point. While in the current version the skeleton data is not saved during sequence recording, this might be confusing to someone who wants to add that option.

Thanks,

Marek