MarekKowalski/LiveScan3D #28: Low FPS on decent hardware

skymeson, 2018-08-26:
Hi,

I'm getting 2-4 fps on my i7 machine with a 1080Ti. 

Steps I took, 
1. Fresh build using Visual Studio 2017 , 
2. Start the LiveScanServer
3. Start LiveScan Client and connect
4. Show Live from server window

Other kinect apps get 30 fps on same machine. CPU, GPU, and network activity all look normal. Am I missing something? 

A few other issues I noticed, seems like server doesn't fully shutdown after closing. At least lights on Kinect remain on, and I see KinectService running in background. I can close them in the task manager no problem. 

Was able to stream to Unity using your Hololens project which is really cool. But low fps makes it not really usable for me. 

Really cool project though, and tons of potential.  Thanks for posting!

naruniec, 2018-08-28:
Hi,

I'm sure, that you have compiled and run the Debug version instaed of 
the Release version. The difference can be huge :-)

Best,
Jacek

skymeson, 2018-09-03:
100% problem solved! 

That was an easy one! Thanks for your help!

closed 2018-09-03 by skymeson