MarekKowalski/LiveScan3D #29: Symmetry Calibration Issue

mikesmith207, 2018-09-22:
Hey Marek, great project! Having a slight issue with the calibration. I am using two kinects and two markers. I have followed the calibration instructions, but I seem to have the first kinect mirrored so I don't have a full 360 view of a person. The first marker had 0 for all values in the calibration marker panel. I had the translation y value of the 2nd marker set to 180, as any other alteration to the calibration marker panel would render two separate streams, whereas the 180 value on y would seem to render as one stream. I also changed the z value of the translation to equal the length in meters of the box that the markers were on. It seemed that even if I removed that z value and made it 0, it had no bearing on the calibration. Any help would be greatly appreciated!

MarekKowalski, 2018-09-26:
Hi Mike,

I'm not sure what you mean by "having the first kinect mirrored", could you include a screenshot?
I understand that you have two markers on the opposite sides of some object (a box for example). If that is the case, then the rotation of one of the markers should be all 0 while the other should have 180 rotation around the y axis. You mention that you did this for translation, but I imagine this is just an error in the description.

The z having no bearing on the calibration is surprising. Did you calibrate after changing the value of z or before? The changes will only take effect for calibrations done after the value was changed.

Best regards,

Marek

mikesmith207, 2018-09-26:
[image]
[image]

Thanks for the response! So I have attached two images. Those are the views I get when I do a circle around the subject. I expect to see his back from the 2nd Kinect when I go behind the subject, but I see the front again.

mikesmith207, 2018-09-26:
And I also re calibrated after changing the z value.

mikesmith207, 2018-09-26:
[image]

That was the y value I made 180, the translation.

mikesmith207, 2018-09-26:
[image]

This was the y value I changed to 180, the translation row.

MarekKowalski, 2018-10-01:
Hi, 

Sorry for the delayed response. Changing y in translation to 180 just moves the point cloud 180 meters up i.e. out of the field of view. The second image you posted with rotation set to 180 actually seems ok other than the fact that the distance between the front and the back of the person might be too much, not sure from that angle.

Do remember that what you are seeing is a point cloud and that it does not have back side culling (since it is not a mesh). This means that you will see the front of a person if you look at it from behind, unless there is something occluding it.

Does that make sense?

marek