MarekKowalski/LiveScan3D #31: send skeleton data to server?

shyiszcx, 2018-11-16:
Hello Marek

First I wanna thank you for this amazing project.
I notice the software can handle skeleton data now but I didn't see skeleton in the live view window.
I only saw skeleton in the client window.  
Also could you please tell me how to send skeleton data to the server?

Thanks

MarekKowalski, 2018-11-19:
Hi,

Have you tried enabling the skeletons in the Server's settings?

Marek

earlsorensen, 2021-01-19:
Hi guys, I have tried to set the option for skeleton view or body view in the server settings, but each time I set the "body view only" this crashes the Live View screen. Also if the "Skeleton View" is checked then no skeleton is shown in the live view.

any ideas?

MarekKowalski, 2021-01-22:
Hi, is this on the AzureKinect branch? If so, the skeletons are not supported there so far :(
Let me know if it's on the master branch and then we can debug it.

Thanks,

Marek

earlsorensen, 2021-01-22:
Hi Marek, yes it is on the Azure branch. Will you be rolling out a version on the azure branch in the future?CheersEarl

MarekKowalski, 2021-02-01:
Hi, not sure when I'll have access to Azure Kinects again, so at the moment I do not have plans to add that functionality.
Having said that, it should be fairly easy to add it if you want to give it a shot :)