MarekKowalski/LiveScan3D #34: Using LiveScan3D with Realsense D435

akashcastelino, 2018-12-21:
Hi,
I wanted to inquire if LiveScan3D would work with the realsense d400 series cameras?

Thanks and Best Regards

MarekKowalski, 2018-12-23:
Hi,

In order for the app to work with the realsense you would need to implement the [iCapture interface](https://github.com/MarekKowalski/LiveScan3D/blob/master/include/LiveScanClient/iCapture.h) for that sensor. I don't think it is a lot of work, but it would still require the knowledge of basic C++ programming.

Marek

MarekKowalski, 2018-12-31:
Hi,

iCapture is an interface that the application can use to communicate with an RGBD camera. The idea is, that to work with an RGBD camera, you need to implement a class derived from iCapture that will implement all of its functions. An example of such a class is [kinectCapture](https://github.com/MarekKowalski/LiveScan3D/blob/master/include/LiveScanClient/kinectCapture.h), which implements the iCapture inteface for the Kinect v2 depth camera, the implementation of this class is [here](https://github.com/MarekKowalski/LiveScan3D/blob/master/src/LiveScanClient/kinectCapture.cpp).

Marek

MarekKowalski, 2019-01-28:
Hi,

The D435 does not have a body tracking interface, that's true.
You can go around this limitation by never filling in the array of bodies, which is normally used by the Kinect interface. That should not cause any problems.

Thanks,

Marek

predictcare, 2019-04-18:
Hi, did someone managed to implement iCapture.h ?  Thanks!

telemething, 2019-05-14:
<following>

ideasxiang, 2020-02-24:
Is there any guide on how to implement an interface?

ideasxiang, 2020-02-24:
How do I make the interface work with kinect 360?

MarekKowalski, 2020-02-29:
Hi,
I have never worked with Kinect 360, so  I can't really help you with the specifics.
As for an example you can see the implementation for Kinect v2 here: 
https://github.com/MarekKowalski/LiveScan3D/blob/master/src/LiveScanClient/kinectCapture.cpp
There is also an implementation for Azure Kinect, though the interface was slightly modified, you can find it here:
https://github.com/MarekKowalski/LiveScan3D/blob/AzureKinect/src/LiveScanClient/azureKinectCapture.cpp

Marek

ideasxiang, 2020-02-29:
I looked at the files. There isnt a one to one replacement of the functions
eg getbodyframe. What should i do in this case? Thanks

MarekKowalski, 2020-03-09:
Hi,
If you do not need the functionality provided by the functions that are not present, you can return some sort of an empty input. 

Thanks,

Marek