MarekKowalski/LiveScan3D #35: Calibration issue about calibration.txt

TerryHoQvQ, 2019-03-07:
Hi,

Thanks for your nice share first! I am using two kinects and one marker to calibrate, then get the file 'calibration.txt' on the 'Server' computer.I want to know the extrinsics and intrinsics，but I think the file only provides the extrinsics(T,R).I don't know if I understand it wrong. Cloud you tell me the meaning of the parameters inside?
[image]

MarekKowalski, 2019-03-07:
Hi,

You are correct that the file only provides the extrinsics.
The parameters inside are:
 - translation vector
 - rotation matrix
 - id of the marker used for calibration
 - a boolean indicated whether calibration was successful
You can also see the code that loads the file on lines 133-151 [here](https://github.com/MarekKowalski/LiveScan3D/blob/bd969205c226b10f4769488e76f98c6d144c8dc9/src/LiveScanClient/calibration.cpp).

As for the intrinsics you can obtain them from the Kinect SDK.

Thanks,

Marek

TerryHoQvQ, 2019-03-08:
Hi,

Thank you for your reply so quickly.The T,R means the correlation between world coordinates and camera coordinates，right? I have two Kinects that use the same marker to calibrate, but why did I get only one calibration result?

MarekKowalski, 2019-03-12:
The calibration results are saved by the client app, which receives data from only a single Kienct.

closed 2019-03-13 by TerryHoQvQ