MarekKowalski/LiveScan3D #36: Replay saved stream (bin file) from a database

markusham, 2019-04-12:
Hi, great project Marek! I was trying to go about storing the bin file of the recording in a database and then replaying the bin file on the iOS application I made. Since the project is only setup to work with a socket, do you have any suggestions as to how to go about replaying the recorded stream on-demand (such as storing it in a database and calling to download/play it)? Do I need to extract the frames from the bin file and replay them in a readable file format? Any help would be greatly appreciated!

MarekKowalski, 2019-04-16:
Hi,

I'm not sure what the question is exactly. What format does your app expect? Is the bin file stored locally on the device or is the database remote? Please be more precise.

Thanks,

Marek

markusham, 2019-04-17:
Hi Marek, I'm just essentially looking for a way to replay streams and not just stream them. Do I need to port the player into the unity project and store the bin files locally in that project?

MarekKowalski, 2019-04-17:
Hi,

Thanks, I understand your question now. Yes, you would want to move the [code that reads the bin files](https://github.com/MarekKowalski/LiveScan3D/blob/master/LiveScanPlayer/FrameFileReaderBin.cs)  to a Unity project. 

You might have to place the .bin file in the streaming assets directory so that it is copied to the output project for iOS.

Marek