MarekKowalski/LiveScan3D #37: detail steps or video about calibrate

pczebra, 2019-06-04:
nice work you did! i have trouble with calibration, could you give me more details with calibration, or a demo video is perfect.

MarekKowalski, 2019-06-10:
Hi,

Have you read the instructions under the link below?
https://github.com/MarekKowalski/LiveScan3D/blob/master/docs/manual.pdf
If yes, could you specify what you are having issues with exactly?

Marek

pczebra, 2019-06-11:
> Hi,
> 
> Have you read the instructions under the link below?
> https://github.com/MarekKowalski/LiveScan3D/blob/master/docs/manual.pdf
> If yes, could you specify what you are having issues with exactly?
> 
> Marek

i have read this manual,and tried to calibrate with a box. i have got marker coordinates(with red lines) and  kinect coordinates(with green lines) in the live view window, but i didn't get the message:" Calibrated = true". 
And another question bothers me a lot. My computer has i7-7700 CPU, 8G memory with 6G GTX1060 GPU, but the LiveScanClient only has a 1.3 frame rate,  what should i do to enlarge the frame rate.

MarekKowalski, 2019-06-11:
Hi,

I presume you are compiling the code from source, is that correct?
If so, are you compiling in Release (as opposed to Debug) and without the Debugger attached (Ctrl+F5 in Visual Studio)?
When compiling in Debug mode the Calibrated=true does not appear for some reason. it should be fine in Release mode.

Let me know if that helps,

Marek

pczebra, 2019-06-12:
> Hi,
>
> I presume you are compiling the code from source, is that correct?
> If so, are you compiling in Release (as opposed to Debug) and without the Debugger attached (Ctrl+F5 in Visual Studio)?
> When [...]

yes, i compiled the code from source in Debug mode, maybe i should try to compile it in release mode.

thank you for your answering, and thank you for your great work.

JohnnistWangc, 2019-11-12:
- 
[image]

help! 
Why can't I calibrate it?
How do you calibrate the calibration of the parameters successfully?

MarekKowalski, 2019-11-13:
Hey, did you take a look at the manual in the docs directory?