MarekKowalski/LiveScan3D #39: Will you support Azure Kinect?

flufy3d, 2019-08-14:


MarekKowalski, 2019-08-14:
Yes, I hope to have something working soon, perhaps next week.

MarekKowalski, 2019-08-16:
I added a new branch with AzureKinect support. It supports multiple devices connected to the same machine. One instance of LiveScanClient needs to be opened per AzureKinect.

Known issues:
 - Memory starts to increase when more than one client is running on the same machine.
 - Inter-Kinect synchronization is currently not supported
 - No body tracking

MarekKowalski, 2019-09-18:
The memory leak is fixed in [9c219fb](https://github.com/MarekKowalski/LiveScan3D/commit/9c219fbc44d725811af2a5cb02a24afedd16ae68)

MarekKowalski, 2020-08-04:
Merged a PR from Christopher Remde that improves alignment of colour and depth data:
https://github.com/MarekKowalski/LiveScan3D/pull/48

Clear492g, 2020-12-01:
Can the latest version support multi device synchronization and bone tracking？
Thank you for your effort！

Clear492g, 2020-12-01:
Can the latest version support multi device synchronization and bone tracking
Thank you for your efforts！

MarekKowalski, 2020-12-01:
Christopher made a PR for multi-device synchronization, you can see it here: https://github.com/MarekKowalski/LiveScan3D/pull/49
It seems to be fully functional, but I haven't had a chance to check it out properly yet as I have no device while working from home :(

No bone tracking integration so far, but it shouldn't be very hard to add.