MarekKowalski/LiveScan3D #41: Multiple Azure Kinects save recording problem

strangerStar, 2019-10-31:
Thank for share your project , it is so great. I have a problem when i trying to recording the point cloud from multiple Azure Kinects,  The cameras were  calibrated and  can show in live view ,and enable merge scans in setting, After i recording the point clouds, i check the file in MeshLab , only one camera point cloud was saved .  Is it the bug ? or something i m missing?

MarekKowalski, 2019-11-08:
In settings you can check to either save separate files for each Kinect or to save them all into a single file. Have you tried playing with that?

Marek

strangerStar, 2019-11-08:
Oh , I haven't check the setting option before, i let me try.  Thank you.

spike3d, 2024-01-15:
Due to the end-of-life (EOL) status of Azure Kinect, which is no longer maintained, Orbbec company provides Femto Mega and Femto Bolt as replacements that are secure and compatible with the Azure Kinect interface. I have noticed that many former users have been using the pyKinectAzure repository. We have forked this repository and made some simple adaptations to support Femto Mega and Femto Bolt.

1、Femto Mega
Femto Mega is a programmable multi-mode Depth and RGB camera with real-time streaming of processed images over ethernet or USB connections. The camera uses Microsoft’s industry proven ToF technology and the NVIDIA® Jetson™ platform to deliver a comprehensive Depth and RGB vision platform for computer vision and AI developers. With a wide field of view (FOV) and high resolution depth sensor, it’s ideal for robotics, manufacturing, logistics and healthtech applications.

https://www.orbbec.com/femto-mega-document/

Compared to Azure Kinect, it has the following advantages

Lower CPU usage.
Network functionality support.
macOS support.
2、Femto Bolt
Orbbec Femto Bolt is the closest alternate to Microsoft’s Azure Kinect DK ([Microsoft’s Azure Kinect Developer Kit Technology Transfers to Partner Ecosystem](https://techcommunity.microsoft.com/t5/mixed-reality-blog/microsoft-s-azure-kinect-developer-kit-technology-transfers-to/ba-p/3899122)). Developed in collaboration with Microsoft, it uses Microsoft’s advanced iToF sensing technology and has identical operation modes and performance as the Microsoft Azure Kinect DK. Femto Bolt also has a 4K color camera, IMU, and can be operated with a single USB-C connection for power and data.

This document provides a detailed description of the Orbbec Femto Bolt and highlights its differences from the Azure Kinect DK. The information will help developers become familiar with the Femto Bolt and also integrate it into applications originally developed using the Azure Kinect DK.

A detailed comparison of the Femto Bolt with the Azure Kinect DK can be found at [Comparison with Azure Kinect DK](https://www.orbbec.com/documentation/comparison-with-azure-kinect-dk/).

https://www.orbbec.com/femto-bolt-document/

3、GitHub Link
https://github.com/orbbec/pyKinectAzure

OrbbecSDK-K4A-Wrapper:
https://github.com/orbbec/OrbbecSDK-K4A-Wrapper