MarekKowalski/LiveScan3D #42: Q: Camera cannot be successfully calibrated

JohnnistWangc, 2019-11-13:
[image]

MarekKowalski, 2019-11-13:
Hi, did you look at the manual in the docs? Did you print the markers and put them in the scene?

Marek

JohnnistWangc, 2019-11-21:
Thank you very much for your help. The following picture is a reconstruction of the school bag. 
[image]
The effect is good, but there are still some problems. The frame rate collected when rebuilding the dynamic object is too small, causing the image to be ghosted, and whether the slot can be further optimized ?

MarekKowalski, 2019-11-22:
What framerate do you see? (it's the number on top of the live view window).
Are you compiling in Debug or Release mode? If everything is compiled in release you should get 25-30fps.

JohnnistWangc, 2019-11-25:
Yes, but I can only get less than 10fps, and I am using the release version, not the debug version. I checked the "filtering" option.

MarekKowalski, 2019-11-25:
The filtering can slow things down. How fast is it without filtering?
The filtering speed depends on the number of points, so limiting the volume you are using (through settings) will speed things up.
You can also play around with the N neigbours parameter to speed the filtering up.

closed 2019-12-16 by JohnnistWangc