MarekKowalski/LiveScan3D #45: Two Kinects on the same PC

3Duke, 2020-01-12:
Hi,
Liv3D is a great tool, thanks for sharing. I have two Kinects v2 connected at the same computer, and they run with some apps (like “Ipi Recorder 4”) without problems, thanks to UsbDK, a driver that allows two plug two (or more) Kinects 2 on the same PC. 
It would be useful  (to show at demos), not having to work with two computers and two clients….is it possible (and how) to modify the source to acquire from different Kinects and having two instances of LiveClient? I’m not C programmer, only basic stuff. Perhaps it would be slow capturing with one machine, but interesting enough to try it.

Thanks!

MarekKowalski, 2020-01-12:
Hi,
In order to use live scan with that driver one would need to implement the iCapture interface (a c++ class) for that version of kinect sdk. If you want to do that I am happy to help.
However, the Azure Kinect support for multiple sensors connected to the same machine is much better.
There is a branch in this repo that already has that support enabled. It might thus be easier to just get 2 azure kinects.

Marek

3Duke, 2020-01-15:
Hi,
Sure, Azure is the way to go, but unfortunately, we’re not developers, only a public school with a limited budget, and can’t afford new hardware; we have 3 Kinects v2 (and 10 Kinects 1) all of them second hand.

I have poor   knowledge on C programming , I think I couldn’t do that modification; if some  day anybody has the time to do that it would be very useful to a lot of us; but now  with the possibility of working with networked clients I can do some experiments.

Thanks for your great work and  efforts!

padraigfl, 2023-08-04:
Hi, super cool project! Posting in here to mention two things:

1. These days you can get multiple Kinect v2 for next to nothing, and they're relatively low cost to mod for PC (provided you've a sufficient 12v power supply). For the cost of one Azure Kinect you could get a dozen v2 Kinect set up so it would open the project up to a massive number of people if it was possible
2. I'm pretty sure the multiple Kinects per system solutions rely on libfreekinect2 rather than anything to do with the official Microsoft Kinect 2 SDK. Would that mean that this project ultimately wouldn't be very easily adapted to use multiple Kinect? My guess would be yes, but if not I might have some questions to see if I can possibly tackle it (hardly any CPP experience other than some tiny PS Vita mods)

Given the low cost of Kinect 2, would you be able to give an estimate of how many Kinect would be optimal to use simultaneously across multiple computers?

ChristopherRemde, 2023-08-06:
Hey!

I agree with you on all these points, Kinect2s are cheap and still offer a pretty decent quality! 

For these reasons, I'd also like to re-integrate the sensors into the new version of LiveScan I'm working on right now, however this would still take a bit of time. If anybody could code up an interface to the sensor with the other SDK, I'd gladly include this in the new version of LiveScan :)

padraigfl, 2023-08-06:
^ @ChristopherRemde just to verify, are you saying libfreenect2 does currently have the potential to work with this library? I was under the impression the library depends on the skeleton data from the official SDK

I dunno how much I can help with this but I will have a bit of a poke around on libfreenect2 once I've got two kinects modded and working (probably early next month).

ChristopherRemde, 2023-08-06:
The skeleton data is a very optional feature, if your goal is to capture volumetric video, its not needed, and I haven't used it myself in the last few years! I actually removed it for the new version, because the skeleton support from Kinect Azure is not that great.

If you have any progress, let me know!
You can also join our volumetric community discord server, most of the development talk happens there :)
https://discord.gg/cGyFmVhwY