MarekKowalski/LiveScan3D #47: probrem with color saved .bin files

you214, 2020-04-13:
I tried with good this app, but I had a trouble with saved pointlcloud.
When I captured human with Azure Kinect, .ply pointcloud files are collectly generated, but in .bin files human faces saved in blue. 
How can I fixed it?

MarekKowalski, 2020-04-20:
Hi, 
Sorry, I don't fully understand your question, what do you mean saved in blue?

Thanks,

Marek

you214, 2020-04-20:
Hi Marek, thank you for your reply.

Two point clouds are shown below. The left is a .ply file saved as-is, while the right is compressed into .bin and decompressed into .ply.
On the left, the color is preserved, but on the right, the color information seems somehow off. Is there any information changed in the process of compression?
I captured this pic with using the azure-kinect branch.

thanks,

[image]

you214, 2020-04-23:
Recently, I've found that in de-commpresed pointcloud file, only blue value are doubled compared with as-is saved pointcloud file. By reducing this value would avoid the problem, though I suspects there is problem during compdressing process. 
thanks.

MarekKowalski, 2020-04-28:
Hi,
Sorry for being slow with replies. To me it looks like the red and blue channels of the pixel values are swapped. This usually happens when some data is in BGR format but the program expects RGB. 
I suspect that this is due to some difference between azure kinect and kinect v2, but I don't have either with me to check.

Can you tell me what steps you take exactly to get the .ply files? This would help me find the place in code where the problem happens and I could guide you better towards the answers.

Thanks,

Marek

you214, 2020-05-07:
I also apologize for the delay in replying.
As you said, the RGB of the data had been swapped to BGR, and when I swapped the Rch and Bch arrays, it was indeed the original color.
Here are the steps I took to get the .ply file

(1) original .ply files (correct color)
Using LiveScanServer, .ply files were automatically created in the out folder after Start/Stop Recording. The color information is stored in the correct state in this file.

2) de-compressed .ply files(long color)
In the same way, using LiveScanServer, the .bin file is saved in the same directory after Start/Stop Recording. And I read this .bin file with LiveScanPlayer and then save frame, .ply files were saved in the OutPlayer folder.

MarekKowalski, 2020-05-11:
I'll try to fix it once I have access to a Kinect.