MarekKowalski/LiveScan3D #53: LiveScanClient build from source code failed for Azure Kinect.

ly19264, 2021-01-04:
Hi @MarekKowalski,

I am trying to build the binaries for the LiveScanClient for Azure Kinect and it is giving error "Kinect.h" not found. 

Also, when using the pre-compiled binaries with Azure, the LiveScanClient is giving error "Capture Device failed to initialize!"

Could you please suggest how to use Azure Kinect for the project and solve these issues.

MarekKowalski, 2021-01-11:
Hi,

Are you sure you are on the AzureKinect branch, as opposed to the main branch?
The pre-compiled binary is for Kinect v2 only.

Marek

Chris45215, 2021-02-28:
I'm a different user, but I was having an issue building the source code for the Azure build too - I was getting the error "LNK1104 cannot open file 'opencv_world320d.lib'." I confirmed that the file is in the project directory, and I am on the Azure Kinect branch. I'm using Visual Studio 2019 v16.7.4 and ensured I have the correct build tools. 

My copy of the solution does have built versions of LiveScanPlayer and LiveScanServer (I can't recall if I built them, or if I got those from the repo), but I couldn't successfully rebuild them. LiveScanClient and ICP aren't built and wouldn't build. 

On a hunch, I closed Visual Studio, copied the entire file directory, and pasted it to my desktop, and opened this new copy. For whatever reason, it compiled correctly and runs. This may be related to the fact that I originally had the solution buried deep in my file directory - I've noticed that sometimes VS isn't able to access resources that are very deeply nested. Or maybe it just works better the first time the solution is open. But I really don't know why it worked after moving it, these are just my best guesses.