MarekKowalski/LiveScan3D #57: Using AzureKinect, PointCloud is upside down

kinaco17, 2021-05-25:
Hi.

I am running with one Azure Kinect.
The communication with the server went well, but the point cloud displayed in the live scan is upside down, left and right.
In AzureKinectViewer, it appears correctly.

I tried modifying the for statement in LiveScanServer.OpenGLWindow.OnUpdateFrame, but there was no change.
`                    for (int i = 0; i < PointCount; i++)
                    {
                        int index = PointCount - i - 1;
                        VBO[index].R = (byte)Math.Max(0, Math.Min(255, (colors[i * 3] + brightnessModifier)));
                        VBO[index].G = (byte)Math.Max(0, Math.Min(255, (colors[i * 3 + 1] + brightnessModifier)));
                        VBO[index].B = (byte)Math.Max(0, Math.Min(255, (colors[i * 3 + 2] + brightnessModifier)));
                        VBO[index].A = 255;
                        VBO[index].Position.X = vertices[i * 3];
                        VBO[index].Position.Y = vertices[i * 3 + 1];
                        VBO[index].Position.Z = vertices[i * 3 + 2];
                    }
`

Any help would be greatly appreciated.

MarekKowalski, 2021-05-28:
Hi,

Unfortunately I don't have access to an Azure Kinect to try this (working remotely). What you could do is print one of the markers and perform calibration as described in the docs, with the marker rotation around Z axis set to 180 degrees. This should make everything appear right side up.

Marek

VandlJ, 2023-03-10:
@kinaco17 @MarekKowalski Hi, have you solved it somehow?

MarekKowalski, 2023-03-13:
I think the steps I listed in my previous response should resolve this. i.e. setting the rotation of the marker to 180 degrees and doing calibration.

ERICXUCHI, 2023-08-13:
@kinaco17 Hi, I also use Azure Kinect, but unfortunately, I did not compile the `client.exe, player.exe, server.exe` successfully.
I wonder if you would mind giving me some tips for compilation. I have posted an issue with https://github.com/MarekKowalski/LiveScan3D/issues/68
Thank you very much!