MarekKowalski/LiveScan3D #58: .bin file in sync while ply files out of sync for two camera recordings

jessiez0810, 2021-06-08:
Hello,

I encountered a problem while using LiveScan3D with two Kinect Azure cameras. When I play the .bin file in the player the recording shows the two Kinect cameras are in sync like this: https://youtu.be/aFD6q2zI6XU However, when I chose to play .ply sequence of the same recording to play in the player, the two Kinect cameras appear to be out of sync like this: https://youtu.be/Ophyb6914p0 
It appears that all my recordings are like this. I made sure before recording that both cameras were calibrated and temporal sync is enabled. Would you know why this is?

Thank you.

ChristopherRemde, 2021-06-09:
Hi!

Thank you for posting this bug. I created the temporal sync component for Livescan. I'll try to fix this bug, but it may take a few weeks. I think you're also in the volumetric discord, right? I'll keep you updated there, and submit a PR once I could replicate and fix this issue.

jessiez0810, 2021-06-09:
Hi,

Ah I didnt know that was you. Great!

Jessie