MarekKowalski/LiveScan3D #64: color mismatch

ttkrpink, 2022-06-24:
Hi,
I have two Azure Kinect setups. The calibration looks fine, but there is a color mismatch on the edge. I think it's the background color on the edge that causes that. As shown in the picture, the grey background color is picked up by Cam2 shown in the final live view. Is there any setting I can change to make it look better?

[image]

Thanks a lot.

ChristopherRemde, 2022-06-24:
Hey TTkrpink,

this problem is caused by an imperfect projection of the depth camera image onto the color camera image. This is to some degree unavoidable as the depth and color camera do not share the exact same location, but the projection of the Azure Kinect is also known to be not well calibrated https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1058

I'm currently working on a few improvements to Livescan, including a filter for the described problem above. You can try out the newest version by downloading this branch: https://github.com/Elite-Volumetric-Capture-Sqad/LiveScan3D/tree/ServerUIRework , building it, connecting your clients to the server, select a client and then click on the "Individual Client Settings":

[image]

Then check the "Filter depth map" box, increase the filter value to something like 7 or 9. and then apply to all devices.
[image]

This should give you a much cleaner edge.

This branch is however still very much WIP, so some smaller bugs might occur.

ttkrpink, 2022-06-26:
Thanks for your help. 

I cloned the repo ServerUIRework.  I think the "Create New Issue" function is disabled on that repo, so I have to ask questions here: Two missing files from 5 files prevent me from building the repo LiveScanClient. LiveScanServer builds fine.

[image]

Questions I have with the Capture-Squad/LiveScan3D repo are:
1. I saw AudioRecorderClient in the solution, so is there a way I can record audio with LiveScan3D and playback it?
2. If the PLY file is saved, does it mean I can use Blender to modify it, ie, cut-off the first 10s or remove the last 2s, How about audio? Is that how wildcapture.io demo video was made?

Thanks again for your help.

ChristopherRemde, 2022-06-27:
Thank you for reporting this bug, you're right, the files somehow never got commited. I'll fix this this evening and notify you!

Regarding the other questions:

1. The AudioRecorderClient is not yet ready, but Audio support will be added soon, it's one of the things that's next on my list. This may still take a few weeks though! For now, you would need to record audio externally and then sync it manually via a clap for example.

2. The PLY files are saved on a per frame basis, so you can just delete the frames you don't need. The PLY files are also just pointclouds, not meshes I don't know if they can be imported into blender yet. We're also working on  a solution that meshes and textures the output from Livescan, but it will take some time.

Wildcapture.io seems to be made with a very different technology stack, it's more like classical photogrammetry running at 30 frames per second (With a lot more cameras & equipment). So it's not really comparable.

ttkrpink, 2022-07-01:
So, could you please upload the Log.h and Log.cpp files? The server file won't run without the complied client. I really appreciate your help.

ChristopherRemde, 2022-07-01:
Hey, sorry had a few very busy days, it is uploaded now!

ttkrpink, 2022-07-02:
> Hey, sorry had a few very busy days, it is uploaded now!

Thanks. 

However, the build failed again. There is no resources folder in the LiveScanClient folder.

Build started...
1>------ Build started: Project: LiveScanClient, Configuration: Release x64 ------
1>Copying external DLLs
1>F:\LiveScan3D\LiveScanClient\external\libzstd.dll
1>F:\LiveScan3D\LiveScanClient\external\OpenTK.Compatibility.dll
1>F:\LiveScan3D\LiveScanClient\external\OpenTK.dll
1>F:\LiveScan3D\LiveScanClient\external\OpenTK.GLControl.dll
1>4 File(s) copied
1>0 File(s) copied
1>Invalid path
1>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(147,5): error MSB3073: The command "xcopy /y /d  "F:\LiveScan3D\LiveScanClient\external\*.dll" "F:\LiveScan3D\bin\"
1>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(147,5): error MSB3073: xcopy /y /I   "F:\LiveScan3D\LiveScanClient\resources\" "F:\LiveScan3D\bin\resources\"
1>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(147,5): error MSB3073: :VCEnd" exited with code 4.
1>Done building project "KinectClient.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

ChristopherRemde, 2022-07-02:
I'll try to clone and build the Branch on a fresh copy next week and fix all the issues!
Sorry for the inconveniences, the Branch is only a few weeks old and very much WIP at the moment.

ChristopherRemde, 2022-07-16:
Alright, please try to pull the changes and build again! You might need to completly rebuild the whole project so that it will work.
Sorry for the inconvinience, there was some oversight on my end, didn't check in a lot of binary files

ttkrpink, 2022-07-22:
Thank you, the source code is built fine. Server EXE file crashes after the client is connected to it.

ChristopherRemde, 2022-08-08:
Hey, sorry for the long time, but I didn't have a Kinect to try it out until today.
I pulled this on another computer and tried it out, also crashed for me.

However the crash turned out to be simply due to an error while building, please delete your whole bin folder, set Visual Studio to the following build settings (Release & x64): 
[image]

Then rebuild the entire solution:
[image]

And try it out again! Also please be sure to start LiveScanClient.exe not LiveScanClientD.exe!