MarekKowalski/LiveScan3D #65: Problem with Caliberation

SethiShreya, 2023-04-20:
I am using two kinects with one marker it is not being calibrated and the result is two distinct point clouds. I have tried everything the docs said and can see that it is tracking the marker in the depth window but is showing caliberation=false for both markers.
I have added the marker in the settings. Please help me out here as I don't really understand the concept of marker even when i have read all the explanations you have given in other issues.
Also please tell me if i want to use two markers then how to adjust the translation and rotation i mean you said that point cloud is made when markers are set correctly, i want to ask how do we translate the value on the basis of real time measurement or what? Didn't really understand.
Thanks

ChristopherRemde, 2023-04-20:
Hey Sethi!

Which marker have you printed out? And can you upload a screenshot of the marker settings in your program?
If Livescan recognizes your marker, you should see a green or red outline on the marker in the Client preview window.
It would also help if you can show us a picture of your printed marker, sometimes the ink is too reflective/glossy or is too low in contrast.

For using multiple markers:

Imagine you have a world center in your scene, at 0,0,0 (XYZ). Livescan always centers the viewport around this center, and the marker places the content in relation to this center, so it's a good idea to have the marker always close to it, 
The marker offset is always in relation to this world center, If you don't set the transform or rotation of the marker (all values are 0) the marker will be located at the world center.

Now imagine you want to use four markers, arranged like a cube. For the best placement, the world center should be in the middle of the cube. So the offset should be calculated based on the distance of the marker from the center, as well as the rotation offset. I've drawn a small sketch to illustrate this

So let's say this is our coordinate system, XYZ, with the forward axis being on the Z axis. Therefore in the default rotation (0,0,0), our marker faces into the Z direction.

[image]

If we want to now use a cube configuration, and we say our marker width is 20cms, we arrange our markers like this:

[image]

Livescan uses centimeters for translation and euler angles for rotation. Our cube arrangement gives us the following values:

[image]

SethiShreya, 2023-04-25:
preview of settings
[image]

I have used marker=0

It is not recognizing the marker as I could not see any green or red line also calibrated is showing false too. Can you please send me pictures of how you set up the cameras and marker which I can take as a reference to make mine?

SethiShreya, 2023-04-25:
[image]

Printed marker image that i am using

ChristopherRemde, 2023-04-25:
Hm, everything looks fine to me in your settings and the printed marker seems to be good aswell!
Just to be sure, you did press on the "Calibrate" button inside the server windows after you made sure that both color and depth view of the client can see the marker fully?

You might have to experiement with lighting a bit, try to avoid direct sunlight, but also make sure to have some bright artifical lights in your room. The marker should be well-lit.

If you want to, you can also send some screenshots of your client depth and color preview showing the marker (block out the rest of the room with MS Paint or something for privacy), so I can take a look if everythings looking good there!

SethiShreya, 2023-04-25:
[image]
[image]
These are the images well i haven't stick it to the wall because it was not tracking even when i have stick it to the wall

ChristopherRemde, 2023-04-25:
Try gluing the marker onto a very flat piece of cardboard, wood, or something else, which does not bend.

SethiShreya, 2023-04-25:
One marker worked!

SethiShreya, 2023-04-25:
Is ther problem is with translation, dont know?

SethiShreya, 2023-04-25:
Please help me with this too

ChristopherRemde, 2023-04-25:
So several things:

-You should glue your markers to a flat piece of cardboard. If the marker is bended like in your picture, the chances of being recognized are very low and the precision as well

- All markers should have the same size

- Orientation = Rotation, Translation = Position. If you enter 180 into your translation, it means the marker will be 180 meters away from the center.

- If you do this kind of setup with two kinects facing each other, I'd recommend glueing both markers onto the opposite sides of a piece of cardboard and holding in the middle of the two kinects. This way, you only need to set the Y Rotation/Orientation for one marker to 180 and leave all other values at 0.

SethiShreya, 2023-04-25:
Thank you for all the responses, i am now able to track two markers, 
One last thing i want to ask is how are you setting up your environment for 4 kinects in demo video? I mean how many markers are you using and how are you arranging it?

SethiShreya, 2023-04-30:
Hi, Are you done writing the blog as i was waiting for it to execute with 4 kinect devices?

ChristopherRemde, 2023-04-30:
Hey! Sorry I got sick until yesterday ago and could not finish it yet. I'll have it done in the middle of the week I think!

SethiShreya, 2023-05-05:
Hi, Are you done writing the blog?

ChristopherRemde, 2023-05-08:
I've just finished writing the documentation on the Calibration process, you can view it here: https://github.com/Elite-Volumetric-Capture-Sqad/LiveScan3D/blob/development/docs/User%20Guide/Calibration.md

SethiShreya, 2023-05-09:
Thanks for the clear docs, I have a question, Well I want to ask if my setup is like I want to capture a person sitting on the chair and using a 4-cube marker setup then how will I place the cube to get all the sides of the person sitting on the chair and marker are readily captured by Kinect.

ChristopherRemde, 2023-05-09:
I'd place the cube on the chair, but elevate it a bit with some boxes so that the backrest doesn't cover anything on the cube. Otherwise I'd arrange the kinects like in the graphic: https://github.com/Elite-Volumetric-Capture-Sqad/LiveScan3D/raw/development/docs/User%20Guide/images/Calibration/4_Marker_Setup.png

SethiShreya, 2023-06-01:
Hello
Thank you for all your support i am able to track 4 kinects, but i want to ask you one thing, i have 4 kinect dks and i am trying to connect all the four with daisy configuration but it wont work until i attach all the kinects to separate laptops and connect livescan client, So wont it work if i connect all the kinects with one computer or two?

ChristopherRemde, 2023-06-01:
Hi! What specifially doesn't work? Do you get an error message? It should be able to run all kinects on one PC, that is in fact what I've been doing most of the time.

SethiShreya, 2023-06-01:
Basically whrn i connect one kinect to the laptop and other to power source and that kinect and coonect to the server i can only see one kinect connected but when i connect other to other laptops and then server then only i can see other kinects and master subordinate relationship.

ChristopherRemde, 2023-06-01:
Apologies that sentence was a bit hard to understand for me. Can you describe the process again but each step after the other, like this?

1. Connect first Kinect to Laptop
2. Connect second Kinect to Laptop
3.  Open two Livescan Clients on Laptop
...
and so on?

SethiShreya, 2023-06-01:
Sure
1. I connect first kinect i.e. master to server laptop
2. I connect other kinects to the master kinect as daisy configuration and didn't connect subordinate kinects to any laptop but only power and to each other
3. I only connect one client with master kinect to server
4. And enabled the temporary sync but can only see one kinect 
5. I experimented by connecting other subordinate kinects to connect with laptops and open livescan client for all and connect to server then only i can see other subordinated kinects.
So i just wanted to ask if it is possible to use only one kinect for all the 4 kinects?

songqiming97, 2023-06-23:
Hello, thank you for your reply, I have read the document you thanked, but the calibration is still successful. I have two Kinectv2 cameras, and put them face to face, the marks are placed in the middle, and the marks 0 and 1 are used, and the mark Y of 1 is turned to fill in 180°, and the other marks are 0, and the correction also appears green box, but it is still false

ChristopherRemde, 2023-06-23:
This means the color camera can see the marker, but the depth camera can not. This is usually caused by the marker being too far away or too close to the sensor. Try moving the marker around. Sometimes this issue can also be caused by too much IR-light in the enviroment.

songqiming97, 2023-06-24:
你好，感谢你的回复，请问也是通过；移动来让深度相机中队标记图案显示绿色轮廓吗？

songqiming97, 2023-06-24:
Hello, thank you for your reply. Do you mean that I should also display the marked green outline in the depth image?
[image]
[image]
As shown in the image above, shadows can only be roughly shown

songqiming97, 2023-06-24:
[image]
The above picture is the image I clicked show live with two KinectV2s to view. I have a question: Why is the Marks displayed by kinectv2 at each end the Marks at the back of the display board? Is it because my board is too thin, I pasted marks of 0 and 1 on the front and back of a cardboard

songqiming97, 2023-07-07:
According to your requirements, I pasted Post-it notes on the ground to prevent the ground from being light resistant, but the result of Calibrate still shows false。
[image]
[image]
[image]

songqiming97, 2023-07-07:
[image]
I placed the two kinect positions opposite each other, the LiveScanServer connected Kinect face is tag 0, and the second Kinect V2 is shot tag 1

ChristopherRemde, 2023-07-07:
In the last post it looks like your setup is correctly calibrated? So you managed to fix the problem?

songqiming97, 2023-07-08:
The point cloud data I photographed was Calibrated, but the LiveScanServer window showed Calibrated = false

ChristopherRemde, 2023-07-08:
That's a bit odd for sure, but if the pointcloud is correctly calibrated, the calibration is successful!

I'm not too sure why this state isn't reflected in the server window, but you can assume that this is just a bug for now.

songqiming97, 2023-07-11:
Hello, thank you for your reply. May I ask why my camera is at the same height, but the position of the green coordinate (Kinect V2 camera) in the show live screen is not at the same height, and the reconstructed data is slightly inclined。
[image]

ChristopherRemde, 2023-07-11:
Could be mutiple reasons:

- Are you sure that both kinects do have the exact same height and angle in the real world?
- Are the two markers precisely aligned?
- Are the markers tightly attached to the cardboard, and is the cardboard and marker flat and not bend?
- Did you make sure that the kinects haven't moved since calibrating?
- Two kinects facing each other like this can introduce interferences, because the lasers of both kinects shoot each other in the camera

songqiming97, 2023-11-15:
Hello, can you send the document about correction? It seems that the link is invalid

ChristopherRemde, 2023-11-15:
Ah yes, I'm currently restructuring to put all the documentation on a website, so the file has moved here: https://github.com/BuildingVolumes/LiveScan3D/blob/development/docs/Website/content/docs/user-guide/calibration.md

songqiming97, 2023-11-15:
Hello, can I add your wechat or other contact information? I would like to ask you some questions

ChristopherRemde, 2023-11-15:
You can join our discord if you'd like to!
https://discord.gg/7sgwg5Eq

But also feel free to ask question here in the issues/open new issues, in this way other people can search for them if they have the same problem!

songqiming97, 2023-11-15:
OK, I'm in. Thank you

songqiming97, 2023-11-15:
Hello, I would like to ask you about my Calibrate problem,The two Kinectv2 cameras were placed symmetrically, with the mark placed in the center. In the software environment visio stdio 2019, I calibrated that the mark was always shown to be false. The mark was printed in A4 paper size, should it be changed to A3?
[image]

[image]
[image]

ChristopherRemde, 2023-11-15:
It looks like your markers were mirrored on the horizontal axis before printing.
Try mirroring them, after printing they should look like this https://github.com/MarekKowalski/LiveScan3D/blob/master/docs/calibration%20markers/0b.gif, with the single square showing to the left.

songqiming97, 2023-11-16:
Thanks for your patient reply, but I just reprinted it and re-calibrate it, but it still shows false
[image]
[image]

songqiming97, 2023-11-16:
Hello, I have looked up the relevant documentation on corrections and found that the depth map does not match the individual corrections in your documentation, and the interface for setting markers is different from your documentation
[image]

ChristopherRemde, 2023-11-16:
Sorry, yes the documentation is for a new version of LiveScan, which unfortunately doesn't support Kinect V2 anymore. But altough the UI looks different, the functionality should be similar, with the biggest difference being that you manually need to add the markers and marker IDs in your version. Your settings look fine though!

To be honest, I'm really not sure what the problem might be anymore. Your settings look good, the markers are getting correctly detected, (As shown by the green outline) and the depth image also looks good.

You could try to experiement with the light settings a bit (switch the light on/off, or dimm it), and rotate/move the marker around a bit, that sometimes helps.

As a last resort, can you maybe upload a video where you go through the whole process of calibration, from Opening the app to trying calibration a few times?

songqiming97, 2023-11-18:
Hello, I would like to ask how many versions of vs are used for this project?

ChristopherRemde, 2023-11-19:
How many? I'm just using one version, 2022.