MarekKowalski/LiveScan3D #67: Save data in .ply format

simahsa, 2023-05-19:
Is there any way to save the registered reconstructed 3D scene in a .PLY format?
We can have the .ply of each frame but they are not registered. There is only one registered file in .bin format that can be played in LiveScanPlayer but don't know how can we convert that format to .ply

ChristopherRemde, 2023-05-22:
What do you exactly mean by, the ply frames are not registered? Do you mean that you have a single frame for each camera individually, but not the data from all cameras in a single .ply file? In this case, you can just combine the plys from each camera into one frame, with Meshlab for example, as they are still registered, but just output into seperate .plys.

You can also use the LivescanPlayer to open all the bin files, and then check the "Export PLY" button. Now when you play your sequence, it will be exported into the "LiveScanfolder/bin/outPlayer/ folder as .plys

simahsa, 2023-05-22:
I apologize for any confusion caused. I currently have a single Azure
Kinect camera. When I initiate the "start recording" function, it begins
counting the frames. Consequently, it generates a single .bin file that can
be played using LivescanPlayer. Additionally, it creates a folder named
"out" containing individual .ply files for each frame recorded by the
camera. Therefore, if there were 10 frames recorded during the
client/server session, there would be 10 corresponding .ply files alongside
the single .bin file.

While LivescanPlayer enables playback of the .bin file, it lacks an "Export
PLY" button to generate a consolidated .ply file representing the
registration of all the frames in the "out" folder. Although it is possible
to manually perform this registration using MeshLab, I would like to
inquire whether LiveScan3D offers an automated solution for this process.

> Message ID: ***@***.***>
>

ChristopherRemde, 2023-05-22:
Thanks for clearing this up! Just one more questions, when you talk about registration, can you describe a bit further what you mean? Registration, at least in the computer vision field, often refers to aligning two 3D-models, or pointclouds, that are made from the same scene, but from different viewports/cameras (https://en.wikipedia.org/wiki/Point-set_registration). But you said that you use only one camera, so registration in that sense would not be needed.

simahsa, 2023-05-22:
You're absolutely right! I require registration because I move the camera
while recording so the camera's position changes in each frame. Considering
this, it might be more advantageous for me to utilize SLAM (Simultaneous
Localization and Mapping) techniques.

ChristopherRemde, 2023-05-22:
Ah I see! Yes that is not a usecase that is covered by LiveScan3D, in this software the cameras are meant to be static. But there are other solutions for this, most prominently KinectFusion (https://www.microsoft.com/en-us/research/wp-content/uploads/2016/02/ismar2011.pdf) which you might want to look into :)

simahsa, 2023-05-22:
I'll take a look. Thanks for all your help!