MarekKowalski/LiveScan3D #68: How compile .exe for Azure Kinect?

ERICXUCHI, 2023-08-13:
Hi, everyone. I tried to compile the .exe for Azure Kinect with visual studio 2019 but failed. The error outputs are like below:

`2>C:\Users\EricXu\Documents\LiveScan3D\LiveScanClient\KinectClient.vcxproj(202,5): error : This project references NuGet package(s) that are missing on this computer. Use NuGet Package Restore to download them.  For more information, see http://go.microsoft.com/fwlink/?LinkID=322105. The missing file is ..\packages\Microsoft.Azure.Kinect.Sensor.1.4.1\build\native\Microsoft.Azure.Kinect.Sensor.targets.`

LiveScanServer: `3>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targets(1217,5): error MSB3644: 找不到 .NETFramework,Version=v4.5 的引用程序集。要解决此问题，请为此框架版本安装开发人员工具包(SDK/目标包)或者重新定向应用程序。可在 https://aka.ms/msbuild/developerpacks 处下载 .NET Framework 开发人员工具包`

LiveScanPlayer: `4>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targets(1217,5): error MSB3644: 找不到 .NETFramework,Version=v4.5 的引用程序集。要解决此问题，请为此框架版本安装开发人员工具包(SDK/目标包)或者重新定向应用程序。可在 https://aka.ms/msbuild/developerpacks 处下载 .NET Framework 开发人员工具包
`

I want to ask if there is someone who has successfully compiled the three .exe for Azure Kinect from source? Or is there any possible alternative solutions for this problem?

Thank you very much!

ChristopherRemde, 2023-08-14:
Hey!

As the error message is saying, you're missing the Nuget Packages for the Azure Kinect. You can either right-click on the solution inside of the Solution Explorer in Visual Studio and then click on "Restore Nuget Packages" or you manually download and install the Azure Kinect Nuget Package inside of Visual Studio yourself. This is the package: https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/

ERICXUCHI, 2023-08-14:
Yeah! I solved this! Thank you very much!

closed 2023-08-14 by ERICXUCHI