MarekKowalski/LiveScan3D #69: Problems about showing .ply in LiveScanPlayer.exe

ERICXUCHI, 2023-08-14:
Hi, team. I have saved the .ply files, but when I started the player and selected the files, the current number is changing from 0 to 50 frame but nothing shows on the screen, should there be a 3D player showed? Or how I can do to solve this problem?

2 Windows 10 + 2 Azure Kinects

the screen is like below:
[image]

ChristopherRemde, 2023-08-15:
Click on the "Show live" button

ERICXUCHI, 2023-08-15:
I have used two Azure Kinects so there is perhaps no .bin file. 

 Are there any alternative solutions? 😣

ChristopherRemde, 2023-08-15:
There should be a .bin file in the LiveScan3D/bin/ folder called recording_XXX.bin after you've recorded a capture. Do you see anything there?

ERICXUCHI, 2023-08-16:
Yeah, I saw it under /bin folder. But the .bin file is also about 1-5 kb, there may be something wrong during recording.

ERICXUCHI, 2023-08-16:
What's more, I cannot see anything when I click 'show live' in the server.exe, even though the FPS ~30.

However, if I did not generate the release version and just click 'run' in the visual studio 2019, the 'show live' runs correctly and can show the images. Do you ever encounter this problem?

ChristopherRemde, 2023-08-16:
I see you started a new issue. Could you close this one?

ERICXUCHI, 2023-08-16:
Yeah, no problem. I found that in this repo "https://github.com/BuildingVolumes/LiveScan3D" it works. Thanks a lot!

closed 2023-08-16 by ERICXUCHI