MarekKowalski/LiveScan3D #70: Show nothing when I click 'show live' in the server.exe

ERICXUCHI, 2023-08-16:
Hi, guys, I cannot see anything when I click 'show live' in the server.exe, even though the FPS ~30.

[image]

However, if I did not generate the release version and just click 'run' in the visual studio 2019, the 'show live' runs correctly and can show the images. Do you ever encounter this problem?

[image]

ERICXUCHI, 2023-08-16:
I found it solves in this repo "https://github.com/BuildingVolumes/LiveScan3D".

ChristopherRemde, 2023-08-16:
Yes that is the new version of LiveScan we're working on right now and will eventually replace this version. However it's still work in progress and probably contains some bugs

ERICXUCHI, 2023-08-16:
Yeah, but it seems that I cannot find the 'issues' section in the new Repository. :(

[image]

ChristopherRemde, 2023-08-16:
Im very sorry, I forgot to enable it, please try again now!

ERICXUCHI, 2023-08-16:
Okay, then I would close this issue. Thanks for your patience and support!