MarekKowalski/LiveScan3D #71: Could we save the depth map image at the same time?

ERICXUCHI, 2023-09-13:
Hi, team, thanks for your effort in this project.

I am using 2 Azure Kinects and I wonder if I could save the depth map image at the same time. Since right now we can only save .ply file, so we wanna save the depth image (like "show depth" in the LiveScanClient) simultaneously.

Or did you leave any API that I can leverage to extend the function?

ChristopherRemde, 2023-09-13:
Hey!

Yes we have this functionality, again over here in the improved version of LiveScan:

https://github.com/BuildingVolumes/LiveScan3D/tree/development

In the server UI, you need to select "Raw Images" instead of pointclouds. But you can't record pointclouds & raw data at the same time. I however have a WIP tool I can share with you, that would allow you to convert the raw data back into pointclouds after the capture!

ERICXUCHI, 2023-09-14:
Yes, thanks for your help. Can you share me with the WIP tool? I have sent a email to you~ 👍

ChristopherRemde, 2023-09-14:
Just if any reader is interested, the tool can be found here: https://github.com/BuildingVolumes/volumetricpipeline/tree/main/DepthProcessingTools/OfflineK4AImageToPointcloud

After downloading, open the solution and install the Azure Kinect Nuget package inside of the project. Then enter the path to your raw Image capture on line 37 in OfflineK4AImageToPointcloud.cpp, all slashes in the path need to be double backward slashes.
Now create a folder called “out” in your entered path and then start the program. You should have .ply files inside of the “out” directory when the program has finished.

ERICXUCHI, 2023-09-14:
👍 👍 👍

closed 2023-09-14 by ERICXUCHI