MarekKowalski/LiveScan3D #73: adding audio

hrhouma, 2024-04-09:
Hello,
Could you provide any guidance or suggestions on how best to synchronize video and images with audio tracks, for a precise alignment between visual and auditory elements.
Thank you for your support and for developing such a versatile tool. 
I look forward to any recommendations you might have.

ChristopherRemde, 2024-04-09:
Hi!

This is indeed somewhat unsolved at the moment. If you take a volumetric video with the Azure Kinects and Temporal Hardware Sync enabled, you should get a video that runs mostly perfectly on time (no dropped frames). If you do some "analoge" syncing, like clapping your hands, and then adjusting the audio clip manually, you should have pretty nicely synced recordings. Especially for short clips (<3 minutes) this should work reasonably well.

Could this maybe help you?

hrhouma, 2024-04-09:
I require synchronized color images, depth images, and sound. At the moment, point cloud data isn't necessary. Is this achievable?

ChristopherRemde, 2024-04-12:
Hey! Yes that's exactly what I meant :)

Official audio support will come eventually, but I wouldn't count on it in the next time!