MarekKowalski/LiveScan3D #75: Deployed in a Linux environment

zhuozhuoyueguang, 2025-01-17:
Hello author, what should I pay attention to if I want to deploy this project in a Linux environment? It seems that visual studio 2013 cannot be installed on Linux

ChristopherRemde, 2025-01-17:
Hey! 

Unfortunately, deployment in a Linux environment is not possible, as the GUI Framework used is only compatible with Windows. 

Currently there unfortunately also no plans to bring this software to Linux, so you may have to resort to a VM at the moment. 

Best regards: 

Christopher

zhuozhuoyueguang, 2025-01-20:
Is it possible to deploy in Linux without using a GUI framework and just generate a point cloud map? What I want to know is whether the plugins and environments compiled in the project can find a replacement under Linux. Thanks for the restoration

ChristopherRemde, 2025-01-20:
Kind of yes, the underlying libraries (OpenCV, Azure Kinect SDK) all run on linux as well, but the app would need a rather large rewrite to be able to do this! I don't have the time for this at the moment, but if you want to tackle this, please let me know and we can make a plan!

zhuozhuoyueguang, 2025-01-20:
Yes, I would like to deploy the project in a Linux environment and implement dynamic point cloud fusion, where one camera is fixed and the other camera is moving, so that it can be placed at the traffic light intersection to eliminate the blind spot of the vehicle driver

ChristopherRemde, 2025-01-20:
In this case please note that moving cameras are not supported in this project. For the pointcloud fusion to work, all cameras need to be static, and manually calibrated each time a camera has been moved! So I'm not sure if this tool is the right starting point for you.

zhuozhuoyueguang, 2025-01-20:
Do you think it is difficult and achievable to fix one camera and move the other dynamically to fuse the point cloud composition? Thank you again for your prompt reply

ChristopherRemde, 2025-01-20:
It is certainly more difficult but there are approaches, it is worth looking into papers like Kinect fusion, or RGBD SLAM approaches. I don't spontaneously know of a certain paper that has the exact use case you described!

zhuozhuoyueguang, 2025-03-13:
I want to port this project to Linux, how can I contact you?

ChristopherRemde, 2025-03-13:
Hey! You can send me a mail at the adress in my profile!